chariotsolutions/phonegap-nfc #414: nfc.makeReadOnly() only executes the first line in a then block

TheLazyHatGuy, 2020-06-17:
I use the code below to make tags read-only on android but for some reason, when the read-only is successful I only get the console log and none of the other code runs. Logcat and chrome do not show any error of exceptions

This happens with package version 1.1.1 and capacitor 2.2.0

```js
  private makeTagReadOnly() {
    this.scanningTag = true;

    this.nfcSubscription = this.nfc.addNdefListener(() => {
      console.log('successfully attached ndef listener');
      this.statusMessage = 'Ready to make tag read-only. Please hold your phone up to the tag';
      this.cdr.detectChanges(); // This will make sure that changes to messages are displayed
    }, (err) => {
      console.log('error attaching ndef listener', err);
      this.statusMessage = 'ERR: ' + err;
      this.scanningTag = false;
      this.cdr.detectChanges(); // This will make sure that changes to messages are displayed
      this.nfcError(err);
      this.statusMessage = err;
      this.updateLoading(1);
    }).subscribe(() => {
      this.nfc.makeReadOnly().then(data => {
        console.log('Tag is read-only', data);
        this.scanningTag = false;
        this.statusMessage = 'Tag is now read-only';
      }).catch(err => {
        console.error('Failed to make a tag read-only', err);
        this.scanningTag = false;
        this.nfcError(err);
      }).finally(() => {
        this.unsubscribeNFC();
      });
    });
  }
```

don, 2020-06-18:
This might be a problem with the ionic wrappers. Can you test this with a simple app using just the callbacks? The ionic code is another project. This is also a pain to test since it locks a tag every time it is run.

Looking at the code, it looks like it logs that the tag is read only then unsubscribes. What code isn't running?

TheLazyHatGuy, 2020-06-19:
In the then block only the console.log runs. Setting scanningTag to false and changing the status message don't run at all

don, 2020-06-19:
Do the lines not run or do you not see the result? If you're using Ionic Angular, you probably need something like ngZone to the UI refreshes. Capacitor probably has something similar. 

    this.ngZone.run(() => {
        this.scanningTag = false;
        this.statusMessage = 'Tag is now read-only';
    });

I think this is an app issue rather than a problem with the plugin.

closed 2020-07-09 by don