chariotsolutions/phonegap-nfc #417: How to integrate payment with cordova nfc

mohammedgaus, 2020-06-24:
want to implement the #nfc support for my #cordova application to make #payment using cards Also need the virtual device for testing this on system while developing without mobile or nfc tag Steps needed to complete this

don, 2020-07-09:
Unfortunately you can't do payments.

[Core NFC doesn't support payment-related Application IDs](https://developer.apple.com/documentation/corenfc)

On Android you can use [Host Card Emulation](https://developer.android.com/guide/topics/connectivity/nfc/hce) to get low level access to build apps, but you're not going to have access to secure element. You could build a loyalty card, but not a payment app. This plugin does not support Host Card Emulation, but I have another one that does: [cordova-plugin-hce](https://github.com/don/cordova-plugin-hce).

If you just want to accept payments, you could use a different plugin to do ApplePay or AndroidPay.

closed 2020-07-09 by don

mohammedgaus, 2020-07-14:
Thank you so much @don really appreciated your reply

mohammedgaus, 2020-07-14:
As your @don  suggestion I have used the "cordova-plugin-hce" plugin to integrate the payment support using HCE to do payment with POS machine. And I clone the code from this link https://github.com/don/cordova-hce-demo and tried to run and created the apk file successfully but I dont know this code is in working condition or not
But with current covid lockdown situation I cant test this with real devices, So can you please help me or suggest something which I can use for unit testing as a developer

Thanks