chariotsolutions/phonegap-nfc #429: ISO 15693 tranceive "Tag was lost"

dotbenj, 2020-09-27:
Hi there. I used the plugin on an Ionic application to read an NFC Tag ISO 15693 inner memory with the function "Tranceive" and I have a lot of differents Android devices that return an error called "Tag was lost".

Here is the list of Android devices I tested :

Nexus 5: OK
OnePlus 6: received error "Tag was lost"
Huawai P8L: received error "Tag was lost"
Note 10 Galaxy: OK
Samsung S10e: OK
Oppo: received error "Tag was lost"
Samsung galaxy: received error "Tag was lost"

The chip I'm trying to read from is an ST Microelectronic M24LR04E-R.
Digging into docs about this chip I found an android SDK to read from it. But knowing it's ISO 15693 I thought tranceive function should worked doesn't it ?

I was wandering if anybody had similar issues to know if it come from the NFC Tag itself ?

dotbenj, 2020-10-07:
Just found that if you want to read multiple block the command should start by "02 23" instead of "00 23" it will work with every rfid android compatible devices.

closed 2020-10-07 by dotbenj

dhruv-ff, 2022-02-17:
@dotbenj Hello, Thanks for the information.

```
let writeData = new Uint8Array(4);
                       writeData[0] = 0x02;
                        writeData[1] = 0x23;
                        writeData[2] = 0x00; // Block index
                        writeData[3] = 0x02;
```

I am using this piece of code to retrieve data. What command should i pass to retrieve all blocks of data in one time without sending the block index in a loop.