chariotsolutions/phonegap-nfc #430: nfc.addNdefListener() does not fire .subscribe() function

filipap, 2020-10-28:
Good afternoon, I've been using your plugin to test communication between 2 android devices.  
Here is the code I use to receive the message:
```ts
receiveValue() {
    this.nfc.addNdefListener(ev => console.log('event: ', ev)).subscribe((message) => {
        console.log(message)
    });
}
```

To send the message:
```ts
sendValue() {
      let value = this.ndef.textRecord('Hello World!');
      this.nfc.share([value]);
}
```
I'm using the Ionic Native version that's in the npm registry.
After some testing I noticed 2 problems:
 - when reading a tag the .subscribe() method is never fired after I read the nfc tag.
 - when I was sharing a message with the "share" method and using another tag reader app on the other device, the tag was never detected.

Any suggestions to solve this?

PaintedShepherd, 2021-05-10:
same problem for me. the subscribe function is never fired so i cant read any nfc tags. =(

GilTos75, 2022-03-25:
I had the same problem, the subscribe function was never called for an Ndef tag. In my case the solution was to add the request NFC access in the android manifest as described here

https://developer.android.com/guide/topics/connectivity/nfc/nfc#manifest

I just added the line to the manifest, sync'ed the project and the event fired successfully without changes to the actual code.

<uses-permission android:name="android.permission.NFC" />

archibald-picq, 2022-06-14:
In my case (Android 11), I was missing the one of the geolocation permission (ACCESS_COARSE_LOCATION or ACCESS_FINE_LOCATION)