chariotsolutions/phonegap-nfc #432: Android makeReadOnly doesn't work

Austorius, 2020-11-16:
Hi,
I want to protect my tags after I write a message, but everytime the program reach the makeReadOnly() method I get the message 'Tag can not be made read only'.
Can someone help me?
Message will be set on a tag but never the protection

* prepAndWriteMessage(text: string) {
        const message = [
            this.ndef.textRecord(text)
        ];
        this.nfc.write(message)
            .then(data =>{ this.callback('Tag written')
            this.nfc.makeReadOnly().then(() => {
                console.log("ReadOnly");
            }).catch((err) => {
                console.log(err);
              });
        }, err =>{this.callback('Error')
        console.log(' Err: ', err)
        });
    }*
[image]

closed 2020-11-27 by Austorius

YanShun88, 2023-03-14:
Hello, I have the same problem now, may I ask how you solve it.

Austorius, 2023-03-14:
@YanShun88 
Hi, I solve the problem to set a Tag (ISO 15693) readonly. It is not possible with the Ionic function "makeReadOnly()" because of the specific ISO tag.
I have to manipulate the bytes of the tag. Here is what I did:
`
  private readonly cmdISO15693 = {
    SYSTEM: '2B',
    INVENTORY: '01',
    OPTION_FLAG: '22', // addressed and high-speed
    READ_SINGLE_BLOCK: '20',
    WRITE_SINGLE_BLOCK: '21',
    LOCK_BLOCK: '22'
  };

this.nfc.connect('android.nfc.tech.NfcV', 500)
      .then(
        async () => {
          const blocks = 28; // 28 read by app TagInfo memory size: 28 blocks with 4 bytes per block
          const cmdLock = this.cmdISO15693.OPTION_FLAG  // 0 flag addressed
            + ' ' + this.cmdISO15693.LOCK_BLOCK        // 1 command
            + ' ' + this.tagIdHexArr.join(' ');         // 2-9 tag uid

          let blockError = false;
          let resTransceive: ArrayBuffer;
          for (let i = 0; i < blocks; i++) {
            const block = this.nfc.bytesToHexString([i]).toUpperCase();
            const cmd = cmdLock + ' ' + block;

            try {
              resTransceive = await this.nfc.transceive(cmd);
              try {
                const view = new Uint8Array(resTransceive);
                const resHex = this.nfc.bytesToHexString([...view]);
                console.log(resHex);
              } catch (e) {
                console.log(e);
              }
            } catch (e) {
              blockError = true;
              return;
            }
            if (blockError) {
              return;
            }
          }
          if (blockError) {
            this.notification.push(new Notification(NotificationType.ERROR, 'Error by setting readonly.'));
          }
          this.nfc.close().then(() => console.log('Closed connection'))
            .catch((err) => console.log('Error closing: ', err));
        });
`
I hope this helps you. I know the struggle and the solution to find took me forever. Feel free to ask me if there is anything unclear

Austorius, 2023-03-15:
@YanShun88 Maybe this can help you to find the cmd http://www.emutag.com/iso/14443-3.pdf https://www.puntoflotante.net/TUTORIAL-RFID-ISO-14443A-TAGS-13.56-MHZ.htm
Unfortunately I don't know how I found out. I think it massive google searching and try all possibilities.

I also used the app NFC TagInfo by NXP. This was really usefull because you saw all blocks. If you set the tag readonly you saw it there.
It look like this:
[image]