chariotsolutions/phonegap-nfc #434: How to get 'ndefMessage' object value using 'nfc.readerMode()' function in ionic/typescript

ShivamShokeen, 2020-11-27:
I created a function under which I am calling this code : -

```
this.nfc.readerMode(this.flags).subscribe((tag) => {
this.sucessMessage = tag;
his.scanSuccessfully('Scanned successfully.');
}, (error) => this.errorMessageNFC(error));
```

When NFC tag is scanned I only got this response : -

```
{
  id : [4, 96, 117, 74, -17, 34, -128],
  techTypes : ["android.nfc.tech.IsoDep", "android.nfc.tech.NfcA", "android.nfc.tech.Ndef"]
},
 

Where as it need to give me response like

{
    "isWritable": true,
    "id": [4, 96, 117, 74, -17, 34, -128],
    "techTypes": ["android.nfc.tech.IsoDep", "android.nfc.tech.NfcA", "android.nfc.tech.Ndef"],
    "type": "NFC Forum Type 4",
    "canMakeReadOnly": false,
    "maxSize": 2046,
    "ndefMessage": [{
        "id": [],
        "type": [116, 101, 120, 116, 47, 112, 103],
        "payload": [72, 101, 108, 108, 111, 32, 80, 104, 111, 110, 101, 71, 97, 112],
        "tnf": 2
    }]
}

I am using 'phonegap-nfc-npm' plugin.
```

wuilmerj24, 2020-12-07:
@ShivamShokeen Hi. Also happens to me. Could you solve it?

VincentMenzel, 2022-06-21:
@ShivamShokeen This also happened to me. Any updates?