chariotsolutions/phonegap-nfc #437: [TAG MIFARE DESFIRE]  When I write on a tag, everything is clear

yannicknya, 2020-12-03:
_NXP MIFARE DESFIRE tags are encrypted by default (DES, 2K3DES, 3K3DES and AES) to secure transmission data._

When I write to an tag `(command : nfc.write)` , nothing can make my data invisible to an application such as: nfc-tool.
I do not know if, I have the possibility of making the data **invisible (encrypted)**.

Can you give me details on how I can solve my problem?

blahblehblah, 2020-12-14:
> _NXP MIFARE DESFIRE tags are encrypted by default (DES, 2K3DES, 3K3DES and AES) to secure transmission data._
>
> When I write to an tag `(command : nfc.write)` , nothing can make my data invisible to [...]

If you're referring to iOS, check this pull request > https://github.com/chariotsolutions/phonegap-nfc/pull/424

yannicknya, 2020-12-14:
I use Android

yannicknya, 2020-12-22:
Its more important, Do you have a solution?