chariotsolutions/phonegap-nfc #443: Tag was lost

Amanda1995pig, 2021-02-01:
Hello,
When I write Nfcv Tag with nfc.transceive ,It returns "Tag was lost".
This is my code:
------------------------------start------------------------
async testWrite() {
   await this.nfc.connect('android.nfc.tech.NfcV')
    this.nfc.transceive('90 5A 00 00 03 AA AA AA 00').then(
      response => console.log(response),
      error => console.log( error)
    )
  }

-----------------------------end----------------------------
I don't know what went wrong.Can you give an example on how to use nfc.transceive() if possible.Thanks a lot

dhruv-ff, 2022-02-17:
@Amanda1995pig Which command should i use to get all the bytes from my NfcV device.? Like the one you have used (90 5A 00 00 03 AA AA AA 00).

carlo318, 2024-12-30:
Hello, I also get every time the error "Tag was lost", in NfcA mode.
But are you sure that the HEX command in string mode must be sent with a space between the hex values?
By the way I get the same error with or without the space.

`
nfc.connect('android.nfc.tech.IsoDep', 100).then(
        () => {
            // Authenticate AES
            var apduCommand = '90AF00000001';				
            nfc.transceive(apduCommand).then(
                (response) => { console.log("Transceive response: " + util.arrayBufferToHexString(response));                },
                (error) =>  {   console.error("Transceive error:", error);                }
            );
        },
        (error) => console.log('NFC Connection failed', error)
    );
`

Any idea?

NPIControls, 2025-01-15:
Hi, I am facing the same issue. Did anyone manage to find a resolution?