chariotsolutions/phonegap-nfc #450: Error ios 14.5 - 'new' is unavaible

Neorichi, 2021-04-29:
Problem and solution
Line 128:  self.nfcSession = [[NFCTagReaderSession **new**] change to  self.nfcSession = [[NFCTagReaderSession **alloc**]
Same into Lines 134, 309, 314 and 322

Thanks

Amadorval75, 2021-04-29:
Same Here, 5 Errors on the NfcPlugin.m file 'new' is unavailable

jaimeurzua, 2021-04-29:
Wonder! Thanks 
I can confirm that the change work perfect ...

geraldapeoples, 2021-04-29:
Thanks for posting this ... the fix works for me too!

danielsigl, 2021-05-01:
Thank you very much! It worked for me too!

BlueFlowSystems, 2021-05-08:
Thank you so much for the solution. It works!

webtimo-de, 2021-05-08:
Thank you, you save my life xD

sbeka, 2021-05-11:
Thank you 3000! 😭

marioshtika, 2021-09-24:
New update: I am not having this problem after updating to Xcode 13.

kamal7jan, 2023-09-28:
Thanks!, it worked for me

Ciriak, 2024-07-01:
So, is this gonna be merged one day ? @chariotsolutions