chariotsolutions/phonegap-nfc #452: XCODE 12.5 throwing error: new is unavailable

dhartl2311, 2021-05-31:
Hello,
i'm using phonegap on my capacitor app. After i updated xcode to 12.5 i'm getting the error 'new' is unavailable. Please find attached screenshots where xcode is throwing this error:

[image]

[image]

loredanKon, 2021-06-01:
Take a look to this #450

marioshtika, 2021-08-20:
Hello @don 

this error `new is unavailable` is a simple but major braking error on Xcode 12.5 and makes this plugin unusable as it also creates a lot of other repositories with the same simple fix.

I can see that there are already 2 pull requests waiting for approval:

- https://github.com/chariotsolutions/phonegap-nfc/pull/451
- https://github.com/chariotsolutions/phonegap-nfc/pull/453

Do you think you can merge the fix, or even commit a fix yourself?

Also, I would like to ask you if you don't have time to fix bugs, implement new features, review pull requests, or merge pull requests. I can help you with that.
We can find a way to collaborate, by either taking over the repository or helping you handle the first steps of answering the community's questions, reviewing the code from the pull requests, suggesting the right changes, and then you can just merge them.

Since I haven't seen you respond to any of the questions the community is asking, a quick answer (👍  or a pm) would be great, so that we can know if this is an active repository or not.

marioshtika, 2021-09-24:
New update: I am not having this problem after updating to Xcode 13.