chariotsolutions/phonegap-nfc #454: Ionic NFC Support is NOT available on IOS

AmaurySchmitt, 2021-06-26:
Hello,

I have a big problem with the NFC (I use Ionic & Capacitor), indeed despite having uninstalled and reinstalled it still gives me the same error when I compile it on an IOS emulator: 
`[CoreNFC] 00000002 822651e0 -[NFCHardwareManager areFeaturesSupported:outError:]:154  XPC Error: Error Domain=NSCocoaErrorDomain Code=4099 "The connection to service on pid 0 named com.apple.nfcd.service.corenfc was invalidated from this process." UserInfo={NSDebugDescription=The connection to service on pid 0 named com.apple.nfcd.service.corenfc was invalidated from this process.`

I don't understand this error, I correctly add the right line on Info.plist to say that I am using NFC : 
[image]

**Ionic v5.33.1**
**Angular v12.1.0**
**Capacitor v1.2.0**

guizmo51, 2021-06-29:
Do you check and enable "NFC" in the signing and capabilities area ? (Don't forget to check for release and debug mode)

leonmeka, 2021-06-29:
I have the same issue, even though NFC capabilities and Privacy Usage Description was added...

AmaurySchmitt, 2021-06-29:
> Do you check and enable "NFC" in the signing and capabilities area ? (Don't forget to check for release and debug mode)

Yes

AmaurySchmitt, 2021-06-29:
> Do you check and enable "NFC" in the signing and capabilities area ? (Don't forget to check for release and debug mode)

Yes but always the same problem, idk why 🥲

guizmo51, 2021-06-29:
Do you try to upgrade CapacitorJS to V2 ? 
Do you try your App on different devices ? Or only simulator ? (NFC doesn't work on Simulator)

AmaurySchmitt, 2021-06-30:
Yes I'm currently in v2.4 for Capacitor, however the NFC work's fine with Android but doesn't work with IOS. Maybe, I need to have an account Apple Developer for unlock the functionality ?

xyboox, 2021-08-20:
>  Maybe, I need to have an account Apple Developer for unlock the functionality ?

I wonder how did you manage to develop anything for ios until now without an apple dev. account!?

helllamer, 2021-09-06:
In your `config.xml` (or possibly in `plugin.xml`):
```
    <platform name="ios">
        <config-file target="*.entitlements" parent="com.apple.developer.nfc.readersession.formats">
            <array>
                <string>NDEF</string>
                <string>TAG</string>
            </array>
        </config-file>
    </platform>
```
From [here](https://stackoverflow.com/a/56691555).

mtrabobank, 2023-01-22:
Also you need to do some setup in the Developer portal.

check the first step here: [https://github.com/revtel/react-native-nfc-manager/blob/HEAD/setup.md](url)