chariotsolutions/phonegap-nfc #456: NDEF_PUSH_DISABLED Cordova android 10.0.1 Android 11

clarklight, 2021-09-13:
I am trying to use the .share() function, but in the error callback it is showing NDEF_PUSH_DISABLED. 

Do i need to set anything in the config to enable this? 

I am building on Cordova android 10.0.1 testing on Android 11 device, the android beam function is removed, so there is no option in the setting page in the device or popup to ask user for permission. 

Thanks!

clarklight, 2021-09-13:
For anyone facing this issue, this will save you sometime. After API level 29, the feature is deprecated, and they are slowing moving away from NDEF file transfer method. 

https://developer.android.com/reference/android/nfc/NfcAdapter

boolean | isNdefPushEnabled()This method was deprecated in API level 29. this feature is deprecated. File sharing can work using other technology like Bluetooth.

SamuelMichelRiad, 2022-05-24:
Any workarounds?

clarklight, 2022-05-24:
@SamuelMichelRiad In that documentation, they suggest people to switch over to use Bluetooth for data transfer. Which was the work around i did in the end.