chariotsolutions/phonegap-nfc #473: NFC scanning not working IOS

Master244, 2022-07-08:
Hi,

I have the plugin setup properly it's loading, giving no errors tried also with different imports 

import {NFC} from "@awesome-cordova-plugins/nfc/ngx";
import { NFC, Ndef } from '@ionic-native/nfc';

both give no errors but the same result. 

the result is that NFC scanning session is started and I see the UI Ready to Scan. 

I first thought it was my NFC tag that was not working but then I tried scanning it within shortcuts -> new automation -> Scan NFC tag and it worked instantly.

This is one a Iphone 13 pro on 15.4.1

We are using this plugin for a client, so hopefully this seems familiar to someone.

dilo99, 2022-07-27:
I don't know why, but the same happens on android version 12 and up. the reading screen appears, it also makes the reading sound, but no event is triggered

dilo99, 2022-08-04:
@Master244 hi, you solved? does anyone else have the same problem on android 12 SDK?

Master244, 2022-08-05:
I was not able to fix it. Still want to fix it tho other options are dropping this library a premium one with proper support since it's not stable.

dilo99, 2022-08-05:
> I was not able to fix it. Still want to fix it tho other options are dropping this library for a premium one with proper support.

@Master244 ok but there is another library to use NFC on Cordova? With a proper support...

Master244, 2022-08-05:
There are many yea but I want to contribute to an open source community especially for these kind of multi applicable components projects should be available to the open source community.

dilo99, 2022-08-05:
Yes, is great, but if it will not be supported anymore.... I'll move to another plugin

Master244, 2022-08-05:
People are working on it, just not at the pace we would like to see :)