chariotsolutions/phonegap-nfc #476: Android 12 SDK not fire scan event

dilo99, 2022-08-04:
Hi, i setup properly the plugin, but when i build with SDK 31 or 32, the listener start correctly, th event of the bond of the listener correctly trigger, but no event trigger the onSuccess scan function...
the phone make the scan sound but no event is triggered
the sound of the scan is different from the phone default sound, so the sound i know that is from the plugin and not from the phone system listener, but no event is triggered!!
why?

Master244, 2022-08-05:
Same issues on Iphone and Android

Paradox7208, 2022-08-25:
After making my own fork of this project and a bit of trial and error, I managed to get my Android 12 SDK firing the scan event. For reasons unknow to me as I am not an Android developer (I rely on Cordova), you need to set the intent flag to MUTABLE instead of IMMUTABLE. 

If you read this section of the Android PendingIntent reference, Android used to treat all implicit PendingIntent references as MUTABLE. So I decided to explicity set that flag myself to replicate how the intent used to be created, and now my app scans NFC tags again.

I hope this works for you too!

[https://developer.android.com/reference/android/app/PendingIntent#FLAG_MUTABLE](https://developer.android.com/reference/android/app/PendingIntent#FLAG_MUTABLE)
[image]

My repo that has the intent flag update which I will submit a pull request for.
[https://github.com/Paradox7208/phonegap-nfc](https://github.com/Paradox7208/phonegap-nfc)

Shadowsith, 2022-08-29:
@Paradox7208  thx for your fork.
I tried it out and it works well for Android 12 (codova android@11.0.0, SDK 32)