chariotsolutions/phonegap-nfc #478: Ionic NFC on Android 12 Crushing the App

bevinGithub, 2022-10-17:
I am trying to get my ionic app to run on Android 12 and all the intents are setup correctly but as soon as l do a live reload the APP crushes . If l build it on API 30 the plugin work fine. 

Android API 31 is having some issues with a number of plugins, your assistance will be much appreciated.

**Environment**
Ionic:

   Ionic CLI                     : 6.20.1 (C:\Users\Tigere Bervin\AppData\Roaming\npm\node_modules\@ionic\cli)
   Ionic Framework               : @ionic/angular 6.2.2
   @angular-devkit/build-angular : 13.0.4
   @angular-devkit/schematics    : 13.0.4
   @angular/cli                  : 13.0.4
   @ionic/angular-toolkit        : 5.0.3

Cordova:

   Cordova CLI       : 11.0.0
   Cordova Platforms : android 10.1.2
   Cordova Plugins   : cordova-plugin-ionic-keyboard 2.2.0, cordova-plugin-ionic-webview 5.0.0, (and 10 other plugins)

Utility:

   cordova-res                          : 0.15.4
   native-run (update available: 1.7.1) : 1.5.0

System:

   NodeJS : v14.15.4 (C:\Program Files\nodejs\node.exe)
   npm    : 6.14.10
   OS     : Windows 10

fjms, 2022-10-25:
Same here:

Error:
`Caused by: java.lang.IllegalArgumentException: foo.bar.app: Targeting S+ (version 31 and above) requires that one of FLAG_IMMUTABLE or FLAG_MUTABLE be specified when creating a PendingIntent.
    Strongly consider using FLAG_IMMUTABLE, only use FLAG_MUTABLE if some functionality depends on the PendingIntent being mutable, e.g. if it needs to be used with inline replies or bubbles.
        at android.app.PendingIntent.checkFlags(PendingIntent.java:375)
        at android.app.PendingIntent.getActivityAsUser(PendingIntent.java:458)
        at android.app.PendingIntent.getActivity(PendingIntent.java:444)
        at android.app.PendingIntent.getActivity(PendingIntent.java:408)
        at com.chariotsolutions.nfc.plugin.NfcPlugin.createPendingIntent(NfcPlugin.java:486)
        at com.chariotsolutions.nfc.plugin.NfcPlugin.startNfc(NfcPlugin.java:534)
        at com.chariotsolutions.nfc.plugin.NfcPlugin.onResume(NfcPlugin.java:814)
        at org.apache.cordova.PluginManager.onResume(PluginManager.java:287)`

bevinGithub, 2022-10-25:
Sorry got it l did manage to install the fork.

ziaurrehmanjutt, 2022-11-03:
> You can apply this pull request:
> 
> https://github.com/chariotsolutions/phonegap-nfc/pull/477/files

Thanks.. 
Its Worked for me also

seikosantana, 2023-05-12:
> Sorry got it l did manage to install the fork.

Would you mind to share how to apply the fix and get the plugin?

Laith-Bzour90, 2023-05-23:
@fjms 
when i add change and build the project the plugin overwrite old func

bevinGithub, 2023-07-11:
hi guys just a question posted it no response. Anyone  has faced this error after writing to 10 tags or so then you start to get this error below:
**NFC Writting Error - Only one tag technology can be connected at a time**

Trexology, 2023-09-04:
I use this command to install with the pull request in

`npm i chariotsolutions/phonegap-nfc#pull/477/head`