chariotsolutions/phonegap-nfc #484: Ionic Android 13 makeReadOnly java.io.IOException

ltivig, 2023-03-06:
After upgrading to Android 13 the **makeReadOnly** either returns "java.io.IOException" error or nothing. It is working without any issue on devices running Android 7 to Android 12.

Any idea will be much appreciated.

Environment:
Ionic:

   Ionic CLI                     : 6.20.8 
   Ionic Framework               : @ionic/angular 5.9.4
   @angular-devkit/build-angular : 0.901.15
   @angular-devkit/schematics    : 9.1.15
   @angular/cli                  : 9.1.15
   @ionic/angular-toolkit        : 2.3.3

Cordova:

   Cordova CLI       : 11.1.0
   Cordova Platforms : android 7.0.0
   Cordova Plugins   : cordova-plugin-ionic-keyboard 2.2.0, cordova-plugin-ionic-webview 4.2.1, (and 23 other plugins)

Utility:

   cordova-res                          : not installed globally
   native-run (update available: 1.7.1) : 1.5.0

System:

   NodeJS : v12.19.0 
   npm    : 6.14.8
   OS     : Windows 10

Ravi2608, 2023-07-11:
Same issue with me also. @ltivig - did you found any solution for this?

ltivig, 2023-07-11:
> Same issue with me also. @ltivig - did you found any solution for this?

No solution yet

jams01, 2024-01-15:
Maybe too late but in folder ./capacitor-cordova-andorid-plugins/java/com/chariotsolutions.nfc.plugin/NfcPlugin.java change line 486
pendingIntent = PendingIntent.getActivity(activity, 0, intent, 0);
it should looks like
pendingIntent = PendingIntent.getActivity(activity, 0, intent, PendingIntent.FLAG_MUTABLE);