chariotsolutions/phonegap-nfc #485: Does not work with Xcode 14.3 and iOS 16.4

srinidhirao, 2023-04-13:
With Xcode 14.3 (14E222b) with iOS 16.4, the nfc plugin fails to compile and shows the following error:

[image]

its mainly because the api semantics has changed in iOS 16.4.

srinidhirao, 2023-04-13:
Please review this pull request which contains the fix: https://github.com/chariotsolutions/phonegap-nfc/pull/486

EYALIN, 2024-07-19:
@srinidhirao  this plugin creator did fantastic work, and we highly appreciate him , but he is not answering for years and unfortunately it is not maintained anymore.
we moved to a maintained plugin (which forked from this one) to here:
https://github.com/EYALIN/community-cordova-plugin-nfc

you can install it as an npm package:
cordova plugin add community-cordova-plugin-nfc --save

I dedicate a considerable amount of my free time to developing and maintaining many cordova plugins for the community (See the list with all my maintained plugins).

srinidhirao, 2024-07-19:
Thanks for maintaining this plugin. I’ll update the plugin and try with Android.

Regards,
Srinidhi

https://github.com/EYALIN/community-cordova-plugin-nfc