chariotsolutions/phonegap-nfc #489: NFC Writting Error

bevinGithub, 2023-06-01:
Hi guys, i have an issue with witting to NFC chips. The writing works fine on a few chips. My client rely heavily on writing the tags at high numbers lets say 30 tags.  When they write to the tags there is a warning message **only one tag technology can be connected at a time** – there is only writing to one tag, there are no other tag near. This happens after about the 4th tag is written too.

Any assistance is much appreciated

Ionic:

   Ionic CLI                     : 6.20.1 (C:\Users\Tigere Bervin\AppData\Roaming\npm\node_modules\@ionic\cli)
   Ionic Framework               : @ionic/angular 6.2.2
   @angular-devkit/build-angular : 13.0.4
   @angular-devkit/schematics    : 13.0.4
   @angular/cli                  : 13.0.4
   @ionic/angular-toolkit        : 5.0.3

Cordova:

   Cordova CLI       : 11.0.0
   Cordova Platforms : android 10.1.2
   Cordova Plugins   : cordova-plugin-ionic-keyboard 2.2.0, cordova-plugin-ionic-webview 5.0.0, (and 14 other plugins)

Utility:

   cordova-res                          : 0.15.4
   native-run (update available: 1.7.2) : 1.5.0

System:

   Android SDK Tools : 26.1.1 (C:\Users\Tigere Bervin\AppData\Local\Android\Sdk)
   NodeJS            : v14.20.0 (C:\Program Files\nodejs\node.exe)
   npm               : 6.14.17
   OS                : Windows 10

bevinGithub, 2023-07-25:
Hi Don.

any feedback or is the plugin still been maintained or supported as we haven't had any feedback on the forum, this is critically urgent.

bommen, 2024-02-26:
@bevinGithub,
Did you manage to find a solution for this issue?

bevinGithub, 2024-05-02:
Hi Bommen.

It's not really a solution, the NFC is buffering after a number of tag reading like 15 reads continuously. So what l ended up doing was just leave the page and unsubscribe the NFC event and go back to the page again after it has cleared the buffer. and resume reading.