chariotsolutions/phonegap-nfc #490: Ionic capacitor targetsdk 33 app crash with plugin is installed without nfc plugin work works ok

adeel399, 2023-07-30:


EYALIN, 2023-08-03:
@adeel399 you can use my Cordova plugin which forked from here and maintained 
https://github.com/EYALIN/community-cordova-plugin-nfc

adeel399, 2023-08-03:
> @adeel399 you can use my Cordova plugin which forked from here and maintained 
> https://github.com/EYALIN/community-cordova-plugin-nfc

So there crashing issue on andoid 12 or above is resolved in that and also the issue plugin not working if we resolve crashing issue

adeel399, 2023-08-03:
I am using capacitor so how i can use that

adeel399, 2023-08-03:
let flags = this.nfc.FLAG_READER_NFC_A | this.nfc.FLAG_READER_NFC_V;
    this.readerMode$ = this.nfc.readerMode(flags).subscribe(
        tag => {
          this.tgval=JSON.stringify(tag);
          alert(JSON.stringify(tag))
        },
        err => {
          alert('Error reading tag'+err)
        }
    )
    
    here is my code to read nfc but on android 13 it does nothing even without your plugin that u suggest to use above..
    plz help what i am doing wrong here.
    i does not even give error alert

EYALIN, 2023-08-04:
@adeel399 in capacitor you can use cordova plugins
https://capacitorjs.com/docs/v2/cordova/using-cordova-plugins

adeel399, 2023-08-04:
> @adeel399 in capacitor you can use cordova plugins
> https://capacitorjs.com/docs/v2/cordova/using-cordova-plugins
> 
> 

It worked but not for android 13

adeel399, 2023-08-04:
> @adeel399 you mean that from Android 13 you cannot add Cordova plugins? or the plugin is not working from Android 13?
> if so, my forked plugin will solve all your issues
> https://github.com/EYALIN/community-cordova-plugin-nfc
> 
> remove the NFC plugin and add this:
> ```
> npm install community-cordova-plugin-nfc
> npx cap sync
> ```
> 
> it is a maintained plugin

I used that npx sync etc before it was not working on android 12 either but now its worked but on android 13 still issue i ll send u error

sonikdesign, 2023-08-04:
I'm here for the same reason, I'm testing on Android 12. App doesn't crash, but no response from nfc.etc methods and events.

adeel399, 2023-08-04:
It gives alert of nfc avilable but on clicking read button nothing shows i have also added alert on error

EYALIN, 2023-08-04:
@sonikdesign @adeel399 did you try the plugin [community-cordova-plugin-nfc](https://github.com/EYALIN/community-cordova-plugin-nfc) ?

adeel399, 2023-08-04:
> @sonikdesign @adeel399 did you try the plugin [community-cordova-plugin-nfc](https://github.com/EYALIN/community-cordova-plugin-nfc) ?

Yes i did i can send u apk if u have android 13 u can check it

EYALIN, 2023-08-04:
@adeel399 great. what is the error in logcat?

adeel399, 2023-08-04:
> @adeel399 great. what is the error in logcat?
> 

Nothing shows

EYALIN, 2023-08-04:
the app cannot crash without anything in the logcat

sonikdesign, 2023-08-04:
@EYALIN your plugin works great. I'm converting an app for Android from Cordova-Android 8.1.0 to 12.0.0. Target Android 13 (API level 33) is around the corner. Phonegap-nfc crashes the app with this new Cordova setup. Now I can read again nfc tags. In the next few hours I hope to test iOS too.

EYALIN, 2023-08-04:
@sonikdesign sound great. follow my plugin so you will be notified on any change.

@adeel399 i think that maybe you didn't install it correctly ?

adeel399, 2023-08-04:
> @sonikdesign sound great. follow my plugin so you will be notified on any change.
> 
> @adeel399 i think that maybe you didn't install it correctly ?

I did install it correctly and for my client its working bcoz he has android 12 but same app on my phone which is android 13 showing nothing ,

EYALIN, 2023-08-04:
@adeel399 if an app is crashing you must have exception in logcat, please try to track it

quiots, 2023-08-21:
Hi @EYALIN, I also have this problem with my Android 13 app with IONIC v6. The application crashes during the NFC scan. I installed your forked community-cordova-plugin-nfc plugin, but I still have the same problem; a SecurityException is thrown. Here's what I see in the logcat:

`FATAL EXCEPTION: pool-3-thread-1
                                                                                                    Process: com.abc.app, PID: 31842
                                                                                                    java.lang.SecurityException: Permission Denial: Tag ( ID: 04 50 A6 3A 2C 5E 81 ) is out of date
                                                                                                    	at android.nfc.Tag.getTagService(Tag.java:388)
                                                                                                    	at android.nfc.tech.Ndef.canMakeReadOnly(Ndef.java:344)
                                                                                                    	at com.chariotsolutions.nfc.plugin.Util.ndefToJSON(Util.java:41)
                                                                                                    	at com.chariotsolutions.nfc.plugin.NfcPlugin.buildNdefJSON(NfcPlugin.java:766)
                                                                                                    	at com.chariotsolutions.nfc.plugin.NfcPlugin.fireNdefEvent(NfcPlugin.java:752)
                                                                                                    	at com.chariotsolutions.nfc.plugin.NfcPlugin.lambda$parseMessage$10(NfcPlugin.java:721)
                                                                                                    	at com.chariotsolutions.nfc.plugin.NfcPlugin.$r8$lambda$Ex2eFqINoQ5KTuUzc8Ttwu00_-8(Unknown Source:0)
                                                                                                    	at com.chariotsolutions.nfc.plugin.NfcPlugin$$ExternalSyntheticLambda0.run(Unknown Source:2)
                                                                                                    	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1137)
                                                                                                    	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:637)
                                                                                                    	at java.lang.Thread.run(Thread.java:1012)`
                                                                                                    	
Does this remind you of anything by any chance? Thanks in advance

PS :  @adeel399 did you finally find a solution for your issue ?

adeel399, 2023-08-21:
Yeah on android side mine is resolve ios stil having issue which @EYALIN working i think on

quiots, 2023-08-21:
Great news

So no more crash on Android 13 for you ? you just installed community-cordova-plugin-nfc? nothing more?

Were you able to see in your logcat if the same exception was raised?

Thank you for your responsiveness @adeel399 !

quiots, 2023-08-23:
Hello again @EYALIN,

Does this remind you of anything by any chance?

`FATAL EXCEPTION: pool-3-thread-1 Process: com.abc.app, PID: 31842 java.lang.SecurityException: Permission Denial: Tag ( ID: 04 50 A6 3A 2C 5E 81 ) is out of date at android.nfc.Tag.getTagService(Tag.java:388) at android.nfc.tech.Ndef.canMakeReadOnly(Ndef.java:344) at com.chariotsolutions.nfc.plugin.Util.ndefToJSON(Util.java:41) at com.chariotsolutions.nfc.plugin.NfcPlugin.buildNdefJSON(NfcPlugin.java:766) at com.chariotsolutions.nfc.plugin.NfcPlugin.fireNdefEvent(NfcPlugin.java:752) at com.chariotsolutions.nfc.plugin.NfcPlugin.lambda$parseMessage$10(NfcPlugin.java:721) at com.chariotsolutions.nfc.plugin.NfcPlugin.$r8$lambda$Ex2eFqINoQ5KTuUzc8Ttwu00_-8(Unknown Source:0) at com.chariotsolutions.nfc.plugin.NfcPlugin$$ExternalSyntheticLambda0.run(Unknown Source:2) at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1137) at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:637) at java.lang.Thread.run(Thread.java:1012)`

I only have this problem with Android 13 with my app under IONIC Capacitor with target SDK 33 : app crashes during scan just after call of nfc.addNdefListener(...).

Thanks in advance if you can give some ideas or any response !

EYALIN, 2023-08-31:
@adeel399 sorry for the delay, i was out of work for a while. 
since the issue you have is with capacitor, it is more challenging for me to test it, since im using Cordova.

minorityuk, 2023-09-22:
@EYALIN  @adeel399 I am using the older version of PhoneGap with capacitor and just wondering how to install your version I have tried npm install community-cordova-plugin-nfc  which installed the package but I can't seem to get access to NFC as it says modules dont exist.

Any steps that I can follow to see if this resolves the issue?

My issue is on Android ISO15693 pops out the tag id in reverse to a nfc pen I have and Apple products.  I want to see if this resolves the issue?

Dev-it-a-dev, 2023-09-27:
+1 having the same issue

NityaSantosh26, 2023-11-21:
Hi @quiots , Did you find any solution for the issue or the exception that is thrown. That is resulting in app crash when the app is launched using nfc chip.