chariotsolutions/phonegap-nfc #493: On ios while reading tag give error 203. "System resources unavailable

adeel399, 2023-08-07:
I already added nfc read usage description but this error came plz help

adeel399, 2023-08-07:
Plz anyone its urgent issue that need attention

adeel399, 2023-08-13:
Anyone ??

EYALIN, 2023-08-13:
@adeel399 feel free to use my plugin, which is forked from this one:
https://github.com/EYALIN/community-cordova-plugin-nfc

it is maintained one

adeel399, 2023-08-13:
> @adeel399 feel free to use my plugin, which is forked from this one:
> https://github.com/EYALIN/community-cordova-plugin-nfc
> 
> it is maintained one

I am using that one brother

EYALIN, 2023-08-13:
@adeel399 sorry. because you opened the issue here, so it was confusing.
i will check it today.

adeel399, 2023-08-13:
> @adeel399 sorry. because you opened the issue here, so it was confusing.
> i will check it today.

Yeah please do

EYALIN, 2023-08-18:
@adeel399 commented in the community plugin:
https://github.com/EYALIN/community-cordova-plugin-nfc/issues/1