chariotsolutions/phonegap-nfc #505: Android Api 34 error Android 14

DalterioRaffaele, 2024-04-04:
Removed Methods

void disableForegroundNdefPush(Activity) |   void enableForegroundNdefPush(Activity, NdefMessage) |   boolean invokeBeam(Activity) |   boolean isNdefPushEnabled() |   void setBeamPushUris(Uri[], Activity) |   void setBeamPushUrisCallback(CreateBeamUrisCallback, Activity) |   void setNdefPushMessage(NdefMessage, Activity) |   void setNdefPushMessageCallback(CreateNdefMessageCallback, Activity) |   void setOnNdefPushCompleteCallback(OnNdefPushCompleteCallback, Activity) |  
Added Methods NfcAntennaInfo [getNfcAntennaInfo](https://developer.android.com/reference/android/nfc/NfcAdapter.html#getNfcAntennaInfo())()

Error build android devices sdk 34

[cordova] nfcAdapter.setNdefPushMessage(p2pMessage, getActivity()); 
[cordova] symbol: method setNdefPushMessage(NdefMessage,Activity) 
[cordova] location: variable nfcAdapter of type NfcAdapter 
[cordova] platforms/android/app/src/main/java/com/chariotsolutions/nfc/plugin/NfcPlugin.java:585: error: cannot find symbol [cordova] } else if (!nfcAdapter.isNdefPushEnabled()) { [cordova] ^ 
[cordova] symbol: method isNdefPushEnabled() [cordova] location: variable nfcAdapter of type NfcAdapter 
[cordova] platforms/android/app/src/main/java/com/chariotsolutions/nfc/plugin/NfcPlugin.java:588: error: cannot find symbol 
[cordova] nfcAdapter.setOnNdefPushCompleteCallback(NfcPlugin.this, getActivity()); [cordova] ^ [cordova] symbol: method setOnNdefPushCompleteCallback(NfcPlugin,Activity) [cordova] location: variable nfcAdapter of type NfcAdapter 
[cordova] platforms/android/app/src/main/java/com/chariotsolutions/nfc/plugin/NfcPlugin.java:590: error: cannot find symbol 
[cordova] nfcAdapter.setBeamPushUris(uris, getActivity()); 
[cordova] ^ [cordova] symbol: method setBeamPushUris(Uri[],Activity) [cordova] location: variable nfcAdapter of type NfcAdapter 
[cordova] /platforms/android/app/src/main/java/com/chariotsolutions/nfc/plugin/NfcPlugin.java:611: error: cannot find symbol 
[cordova] } else if (!nfcAdapter.isNdefPushEnabled()) { 
[cordova] ^ [cordova] symbol: method isNdefPushEnabled() [cordova] location: variable nfcAdapter of type NfcAdapter 
[cordova] platforms/android/app/src/main/java/com/chariotsolutions/nfc/plugin/NfcPlugin.java:614: error: cannot find symbol [cordova] nfcAdapter.setNdefPushMessage(p2pMessage, getActivity()); 
[cordova] symbol: method setNdefPushMessage(NdefMessage,Activity) 
[cordova] location: variable nfcAdapter of type NfcAdapter 
[cordova] platforms/android/app/src/main/java/com/chariotsolutions/nfc/plugin/NfcPlugin.java:615: error: cannot find symbol 
[cordova] nfcAdapter.setOnNdefPushCompleteCallback(NfcPlugin.this, getActivity()); [cordova] ^ 
[cordova] symbol: method setOnNdefPushCompleteCallback(NfcPlugin,Activity) 
[cordova] location: variable nfcAdapter of type NfcAdapter 
[cordova] platforms/android/app/src/main/java/com/chariotsolutions/nfc/plugin/NfcPlugin.java:631: error: cannot find symbol 
[cordova] nfcAdapter.setNdefPushMessage(null, getActivity()); [cordova] ^ 
[cordova] symbol: method setNdefPushMessage(,Activity) [cordova] location: variable nfcAdapter of type NfcAdapter 
[cordova] platforms/android/app/src/main/java/com/chariotsolutions/nfc/plugin/NfcPlugin.java:643: error: cannot find symbol 
[cordova] nfcAdapter.setBeamPushUris(null, getActivity()); [cordova] ^ [cordova] symbol: method setBeamPushUris(,Activity) 
[cordova] location: variable nfcAdapter of type NfcAdapter [cordova] Note: Some input files use or override a deprecated API. 
[cordova] Note: Recompile with -Xlint:deprecation for details.

PleahMaCaka, 2024-04-23:
Same here, is there some solution?

DocBO, 2024-05-10:
Hi, after fiddling around I found a simpl solution that worked for me: I removed all lines which are API 34 incompatible in the java code. 

Actually all the lines with these deprecated functions: void disableForegroundNdefPush(Activity) |   void enableForegroundNdefPush(Activity, NdefMessage) |   boolean invokeBeam(Activity) |   boolean isNdefPushEnabled() |   void setBeamPushUris(Uri[], Activity) |   void setBeamPushUrisCallback(CreateBeamUrisCallback, Activity) |   void setNdefPushMessage(NdefMessage, Activity) |   void setNdefPushMessageCallback(CreateNdefMessageCallback, Activity) |   void setOnNdefPushCompleteCallback(OnNdefPushCompleteCallback, Activity) |  

Since I had in mind a very very simple reader in my app, this worked. I have no idea what these lines were actually doing. 

Also if you wonder how to register functions on Android, the doku was kind of incomplete on that:

this.nfc.addMimeTypeListener("text/demo", this.onNfc).subscribe( (event: any)=>
    { console.log("subscribed.....", event); });
this.nfc.addTagDiscoveredListener(this.onNfc).subscribe( (event: any)=>
      { console.log("subscribed.....", event); });
this.nfc.addNdefListener(this.onNfc).subscribe( (event: any)=>
      { console.log("subscribed.....", event); });

bmarsaud, 2024-05-14:
All these functions are related to Android Beam wich has been deprecated in API 29 and removed in API 34 : https://developer.android.com/about/versions/10/behavior-changes-all#beam-deprecation

It should indeed be removed from this plugin.

JoeyTheCoder, 2024-07-12:
So on my App i'm getting forced to upgrade to SDK 34 otherwise i can't upload any updates to the google console by the end of August. 

Has anyone found a workaround yet to keep phonegap-nfc working in SDK 34?

jramosg, 2024-07-12:
> So on my App i'm getting forced to upgrade to SDK 34 otherwise i can't upload any updates to the google console by the end of August.
> 
> Has anyone found a workaround yet to keep phonegap-nfc working in SDK 34?

same here

bmarsaud, 2024-07-14:
@JoeyTheCoder @jramosg Please see this PR https://github.com/chariotsolutions/phonegap-nfc/pull/508, that has been merged in this forkn[ns0m/cordova-plugin-ns0m-nfc](https://github.com/ns0m/cordova-plugin-ns0m-nfc) and published to NPM : [cordova-plugin-ns0m-nfc](https://www.npmjs.com/package/cordova-plugin-ns0m-nfc).

bevinGithub, 2024-07-15:
> @JoeyTheCoder @jramosg Please see this PR #508, that has been merged in this fork [ns0m/cordova-plugin-ns0m-nfc](https://github.com/ns0m/cordova-plugin-ns0m-nfc) and published to NPM : [cordova-plugin-ns0m-nfc](https://www.npmjs.com/package/cordova-plugin-ns0m-nfc).

This repository works for Android API 34, l have used this. The Reading and Writing to NFC Tags is working. Thanks for the plugin maintenance.

tmwebs, 2024-07-17:
@bmarsaud thank you very much for your contribution

EYALIN, 2024-07-19:
@tmwebs @DalterioRaffaele @bmarsaud @bevinGithub @JoeyTheCoder @jramosg  this plugin creator did fantastic work, and we highly appreciate him , but he is not answering for years and unfortunately it is not maintained anymore.
we moved to a maintained plugin (which forked from this one) to here:
https://github.com/EYALIN/community-cordova-plugin-nfc

******the plugin support SDK 34****

you can install it as an npm package:
cordova plugin add community-cordova-plugin-nfc --save

I dedicate a considerable amount of my free time to developing and maintaining many cordova plugins for the community ([See the list with all my maintained plugins](https://github.com/EYALIN?tab=repositories&q=community&type=&language=&sort=)).

dwettstein, 2024-07-19:
Hello @EYALIN, thanks for your response. Unfortunately, it's (currently) not possible to use your fork together with the Awesome Cordova Plugin wrapper, as you have changed the name and id of the package.
https://github.com/danielsogl/awesome-cordova-plugins/blob/96678cb14c944c25d92fed990ab41c861362feae/src/%40awesome-cordova-plugins/plugins/nfc/index.ts#L92

Additionally your fork is also missing some important fixes, mainly the following two PRs:
- https://github.com/chariotsolutions/phonegap-nfc/pull/460
- https://github.com/chariotsolutions/phonegap-nfc/pull/508

Thanks for your efforts to keep this project alive!

EngageCT, 2024-08-23:
> @tmwebs @DalterioRaffaele @bmarsaud @bevinGithub @JoeyTheCoder @jramosg this plugin creator did fantastic work, and we highly appreciate him , but he is not answering for years and unfortunately it [...]

Hi Eyalin, thank you for the the updating of the NFC plugin. I do have a question when witting to NFC tags, if l write to lets say 20 tags the tags are written fine to 12 tags and after that the tag writing seems to stop writing and write the same tag number like cache bottle neck. Every time l write the tags l have o close the app completely and open it to do the writing again. Is there something wrong or please point me in the right direction. Reading works fine you can read as many as you can writing is the one that's like the cache/buffer is building up. Your assistance will be much appreciated.

bevinGithub, 2024-08-23:
> @tmwebs @DalterioRaffaele @bmarsaud @bevinGithub @JoeyTheCoder @jramosg this plugin creator did fantastic work, and we highly appreciate him , but he is not answering for years and unfortunately it [...]

Hi Eyalin l need your assistance please. I have NFC reading and writing on my app. The reading is fine it's the writing part that is an issue. If l write to lets say 20 tags the tags are written fine from 1 to 12 tags and after that the tag writing seems to stop writing and write the same tag number like cache or the buffer issue. Every time l write the tags l have to close the app completely and open it to do the writing again. Is there something wrong or please point me in the right direction. Reading works fine you can read as many as you can writing is the one that's like the cache/buffer is building up. Your assistance will be much appreciated.
The writing should be smooth regardless of the number of tags you write to.
Any assistance or pointing me in the right direction will be much appreciated.

Writing Function Code below
writeTag(tagData) {
this.writeDate = moment().format('DD-MM-YYYY HH:mm:ss');
this.tag = tagData?.item;
console.log(this.tag);
this.platform.ready().then(() => {
this.writingTag = true;
this.listener = this.nfc.addNdefListener((res: any) => {
console.log('Listener is Active: ' + res);
}, (error) => {
this.presentToast(error);
}).subscribe(() => {
const referenceNumber = this.ndef.textRecord(this.tag?.reference_number);
const date = this.ndef.textRecord(this.writeDate);

this.nfc.write([referenceNumber, date]).then(success => {
  alert(JSON.stringify(success));
  if (success === 'OK') {
    this.listener.unsubscribe();
    this.presentToast('Successfully written to RFID Tag!');
  } else {
    this.presentToast('Could not write to tag sucessfully!');
  }
}).catch(error => {
    if (error === null) {
      this.presentToast('Tag could not be written!');
    }
    if (error === 'Only one Tag Technology can be connected at a time.') {
      this.presentToast('Failed to write to tag sucessfully, please try again!');
    }
  });
}, err => {
this.presentToast(JSON.stringify(err));
});
});
}

My environment information
Ionic:

Ionic CLI : 6.20.1 (C:\Users\Tigere Bervin\AppData\Roaming\npm\node_modules@ionic\cli)
Ionic Framework : @ionic/angular 6.7.5
@angular-devkit/build-angular : 13.0.4
@angular-devkit/schematics : 13.0.4
@angular/cli : 13.0.4
@ionic/angular-toolkit : 5.0.3

Cordova:

Cordova CLI : 11.0.0
Cordova Platforms : android 12.0.1
Cordova Plugins : cordova-plugin-ionic-keyboard 2.2.0, cordova-plugin-ionic-webview 5.0.0, (and 14 other plugins)

Utility:

cordova-res : 0.15.4
native-run (update available: 1.7.3) : 1.7.2

System:

Android SDK Tools : 26.1.1 (C:\Users\Tigere Bervin\AppData\Local\Android\Sdk)
NodeJS : v16.14.0 (C:\Program Files\nodejs\node.exe)
npm : 8.3.1
OS : Windows 10

bagraercan, 2024-09-08:
@EYALIN @bmarsaud unfortunately both of these do not work for me. The app crashes when i set the targetSdkVersion to 34. Any idea or solution?

dwettstein, 2024-09-08:
@bagraercan , we use our own fork (https://github.com/InfosoftSystems/phonegap-nfc) in production and recently released our app to Google Play (with SDK 34 target) and Apple App store with no problems.

bevinGithub, 2024-09-13:
> @bagraercan , we use our own fork (https://github.com/InfosoftSystems/phonegap-nfc) in production and recently released our app to Google Play (with SDK 34 target) and Apple App store with no problems.

Hi [dwettstein](https://github.com/dwettstein), how many tags can you write  to without the nfc buffering. I need to write to 300 Tags on on a site on fire devices detectors.