chariotsolutions/phonegap-nfc #509: NEW Repo for issues and contributions

scr2em, 2024-05-16:
Hi, 

This repo has not been maintainer for over 4 years, I have contacted the owner to add me as a collaborator or transfer ownership but I didn't get any response. 

I have forked it and willing to maintain it and release it to NPM. Probably will be under a different name. 

https://github.com/scr2em/phonegap-nfc

If you have any issue please report it here or if you are a contribute any PR is welcome

EYALIN, 2024-05-16:
@scr2em this plugin creator did fantastic work, and we highly appreciate him , but he is not answering for years and unfortunately it is not maintained anymore. 
we moved to a maintained plugin (which forked from this one) to here:
https://github.com/EYALIN/community-cordova-plugin-nfc

you can install it as an npm package:
cordova plugin add community-cordova-plugin-nfc --save

I dedicate a considerable amount of my free time to developing and maintaining many cordova plugins for the community ([See the list with all my maintained plugins](https://github.com/EYALIN?tab=repositories&q=community&type=&language=&sort=)).

scr2em, 2024-05-16:
@EYALIN Yes, we all highly appreciate him. 

Did you include how to use it with Capacitor?

EYALIN, 2024-05-16:
@scr2em I'm not using a capacitor, and not testing it, but some people that used, it said it works.
but anyway, if not, It will be great if you do PR for it, and I will review and test it.

all I can say is that the plunging maintained and tested on Android and IOS in each new big version

dwettstein, 2024-07-19:
Hi @scr2em, I appreciate your fork and willingness to maintain it!
There is also another fork which has some important fixes: https://github.com/avargaskun/phonegap-nfc/. There is also an open PR in this repo here: https://github.com/chariotsolutions/phonegap-nfc/pull/460. Could you please merge this PR as well?

Thank you!

EYALIN, 2024-07-19:
@dwettstein  this plugin creator did fantastic work, and we highly appreciate him , but he is not answering for years and unfortunately it is not maintained anymore.
we moved to a maintained plugin (which forked from this one) to here:
https://github.com/EYALIN/community-cordova-plugin-nfc

you can install it as an npm package:
cordova plugin add community-cordova-plugin-nfc --save

I dedicate a considerable amount of my free time to developing and maintaining many cordova plugins for the community ([See the list with all my maintained plugins](https://github.com/EYALIN?tab=repositories&q=community&type=&language=&sort=)).

dwettstein, 2024-07-19:
Hello @EYALIN, thanks for your response. Unfortunately, it's not possible to use your fork, as you have changed the name and id of the package. This leads to issues when using the Awesome Cordova Plugin wrapper:
https://github.com/danielsogl/awesome-cordova-plugins/blob/96678cb14c944c25d92fed990ab41c861362feae/src/%40awesome-cordova-plugins/plugins/nfc/index.ts#L92

Additionally your fork is also missing some important fixes, mainly the following two PRs:
- https://github.com/chariotsolutions/phonegap-nfc/pull/460
- https://github.com/chariotsolutions/phonegap-nfc/pull/508

Thanks for your efforts to keep this project alive!

dwettstein, 2024-07-22:
@EYALIN, sorry that my "thumb down" came off as disrespectful. I gave it because I simply asked user scr2em to merge the PR #460. There was no reason to copy your message about your own fork as an answer to me. I felt like this was kind of advertising your own fork, thus I gave a "thumb down".

Of course I appreciate your efforts to the open-source community as well! Thanks.

scr2em, 2024-07-24:
@dwettstein Sure, I'll do that tomorrow

jordodev, 2024-08-07:
Appreciate everyone's contributions!

Just wondering if there is a definitive repo that has all the important changes since this repo stopped being maintained?
Including Android Background launch bug fix, FLAG_MUTABLE fix, ios fixes, Remove Android Beam etc. 

I understand rebasing and merging all of the different forks/changes is a bit of a nightmare but has anyone done it?

dwettstein, 2024-08-07:
Hi @jordodev, I ended up having merged/rebased all important fixes myself in our organisation repo: https://github.com/InfosoftSystems/phonegap-nfc.

However, that's not a definitive repo/fork and not published on NPM. Additionally, it may be also updated without announcement, as we use it in our productive app.

Feel free to use, fork and contribute. Kind regards.